Motsai/neblina-python #17: Flash record and playback commands

acourt, 2015-12-14:
Be able to start flash recording and playback from the python program.

acourt, 2015-12-14:
@osarbishei Care to comment on what exactly you need for this feature?

osarbishei, 2015-12-14:
For details about the packet structures, please refer to the documentation: "_norflash.md"

This is the test routine for the NOR flash recorder:
Step 1 (Start Recording): Send a start recording command, and in response, the Python app will first receive an empty acknowledge packet. It must check for the acknowledge and re-send the command, if it is not acknowledged for some time, e.g., 10ms. Next, the Python app should expect a response packet with the session number that has been created corresponding to the "Start recording command". 
Step 2 (Stream something): Enable a certain motion feature streaming option, e.g., 6-axis IMU streaming. The streaming packets will be sent to the host and they will be recorded in the NOR flash session that has been created.
Step 3 (Stop Recording): Send a stop recording command after some time, e.g., 5 minute. Once again the Python app should expect an acknowledge packet first, which will then be followed by a packet with the session number that has been closed.
Step 4 (Start Playback): Send a playback start command with the session number that has been generated before to open the particular session on Neblina's recorder and start the playback process. The host should receive an empty acknowledge packet from Neblina, which will be followed by another packet with the session number that has been opened for playback. Neblina will also start streaming the content of the recorded session at a high speed rate (around 300Hz). Note that although the acknowledge packet is received before the packet with the session number, it is not guaranteed that the NOR flash contents of the session will be streamed after the acknowledge packet. In other words, the Python app might receive a couple of packets corresponding to the NOR flash contents, before sending the acknowledge packet.
Step 5 (Stop Playback): The Python app can either stop the playback by sending this command, or simply wait until the full contents of the session have been read, and then Neblina will send a COMPLETION packet indicating that the whole sessions has been read and transferred to the host successfully.
Step 6 (Flash Erase): The Python app will send a FlashErase command to Neblina to fully erase the NOR flash. In response, Neblina will send an empty acknowledge packet, and starts the erasing process. This takes a couple of a minutes. After the erasing is done, Neblina will send another packet confirming the completion of the process.

osarbishei, 2015-12-15:
//defined variables
subsystem_nor = 11; //NOR Flash: 0x0B
subsystem_motion = 1;
pkt_type = 2; //command
ctrl_byte_nor = subsystem_nor + pkt_type*32;
ctrl_byte_motion = subsystem_motion + pkt_type*32;
//test procedure (recording mode) for the NOR flash recorder
Step 1: Send start recording command: pkt = [ctrl_byte_nor 0x10 crc 2 ? ? ? ? 1 ? ? ? ? ? ? ? ? ? ? ? 0xC0]
Step 2: Wait until you receive the session created notification: pkt = [subsystem_nor 0x10 crc 2 ? ? ? ? 1 session_nb_LSB session_nb_MSB ? ? ? ? ? ? ? ? ? 0xC0]
Step 3: If after 1s, you did not receive the above packet, then go back to Step 1.
Step 3: Send the enable IMU streaming command: pkt = [ctrl_byte_motion 0x10 crc 3 ? ? ? ? 1 ? ? ? ? ? ? ? ? ? ? ? 0xC0]
Step 4: Wait for the IMU data ACK packet: pkt = [0x21 0x10 crc 3 ? ? ? ? ? ? ? ? ? ? ? ? ? ? ? ? 0xC0]
Step 5: If after 1s, you did not receive the above packet, then go back to Step 3.
Step 6: Wait for some recording time, e.g., 1 minute
Step 7: Send the stop recording command: pkt = [ctrl_byte_nor 0x10 crc 2 ? ? ? ? 0 ? ? ? ? ? ? ? ? ? ? ? 0xC0]
Step 8: Wait until you receive the session recording closed notification: pkt = [subsystem_nor 0x10 crc 2 ? ? ? ? 0 session_nb_LSB session_nb_MSB ? ? ? ? ? ? ? ? ? 0xC0]
Step 9: If after 1s, you did not receive the above packet, then go back to Step 7.
Step 9: Disable IMU data streaming: pkt = [ctrl_byte_motion 0x10 crc 3 ? ? ? ? 0 ? ? ? ? ? ? ? ? ? ? ? 0xC0]
Step 10: Wait for the IMU data ACK packet: pkt = [0x21 0x10 crc 3 ? ? ? ? ? ? ? ? ? ? ? ? ? ? ? ? 0xC0]
Step 11: If after 1s, you did not receive the above packet, then go back to Step 9.

osarbishei, 2015-12-15:
```
//test procedure (playback mode) for the NOR flash recorder
Step 1: Send start playback command for the last recorded session (Session ID = 0xFFFF): pkt = [ctrl_byte_nor 0x10 crc 3 ? ? ? ? 1 0xFF 0xFF ? ? ? ? ? ? ? ? ? 0xC0]
Step 2: Wait until you receive the session opened notification: pkt = [subsystem_nor 0x10 crc 3 ? ? ? ? 1 session_nb_LSB session_nb_MSB ? ? ? ? ? ? ? ? ? 0xC0]
Step 3: If after 5s, you did not receive the above packet, then go back to Step 1.
Step 4: Read a playback packet from the target device
Step 5: If the read packet is the COMPLETION STATUS packet, i.e., pkt = [subsystem_nor 0x10 crc 3 ? ? ? ? 0 ? ? ? ? ? ? ? ? ? ? ? 0xC0], then terminate the program. Otherwise, store the packet, e.g., in a file, and then go back to Step 4. 
```

osarbishei, 2015-12-18:
Default session ID configuration corrected.

commit 501a48f6 (2015-12-17): Added the option to choose the session you want to play back and prints the packets. Fixes #17
closed 2015-12-18 by osarbishei