Motsai/neblina-python #39: Timeout for waiting for device

acourt, 2016-01-14:
One way to implement a timeout is to count the empty debug packets and if you receive a lot of them you can try again. We have to make sure we remove this once the medium of transmission is perfected.

osarbishei, 2016-01-14:
I chose the revision numbers based on JS's comment:
"Byte 0 (Protocol API Release): Currently 0x01
Byte 1 (KL26 Version Major): VERS.Major (Should be 0x01 at the moment)
Byte 2 (KL26 Version Minor): VERS.Minor (Should be 0x02 for Demo 1.0, 0x03 for Demo 1.1)
Byte 4 (KL26 Build Number % 256)
Byte 5 (Nordic Version Major) (Should read 0x01 at the moment)
Byte 6 (Nordic Version Minor) (0x02 for Demo 1.0, 0x03 for Demo 1.1)
Byte 7 (Nordic Build Number % 256)
Bytes 8-20 (Reserved for future use)"

acourt, 2016-01-14:
I think you're on the wrong issue page.

closed 2016-02-09 by acourt