Motsai/neblina-python #57: Plotting the played back 9-axis MARG raw data, temperature and rotation info

osarbishei, 2016-02-24:
Plot info:
- 3-axis Accelerometer data
- 3-axis Magnetometer data
- 3-axis Gyroscope data
- Temperature data
- Speed in km/h, where speed in km/h = rpm * 0.7 * 3.14 * 0.06 (rim diameter is 0.7m)

Also display the total distance traveled, i.e., the last Rotation Count multiplied by 3.14*0.7

acourt, 2016-02-25:
Do the flash sessions create all the same types of packets or will I find packets of different types (mag, imu, etc) in a same flash session when I perform a playback command?

osarbishei, 2016-02-25:
All packet types are in the same recorded session. Maybe we can append the packets to 4 different lists, while playing back from the session to make it simpler. The packet types are IMU, mag, temperature and rotation.

acourt, 2016-02-25:
It's not that bad, I'll just go through the session file and put the packets in different lists.

My script so far:
```python
with open('flashSession1', "rb") as f:
    packetBytes = f.read(20)
    while packetBytes != "":
        packetBytes = f.read(20)
        # Build packet list
        packet = neb.NebResponsePacket(packetBytes)
        subSystem = packet.header.subSystem
        command = packet.header.command
        if subSystem == neb.Subsys_MotionEngine:
        	if(command == MotCmd_MAG_Data):
        		magPacketList.append(packet)
        	elif(command == MotCmd_IMU_Data):
        		imuPacketList.append(packet)
        elif subSystem == neb.Subsys_PowerManagement:
        	if(command == PowCmd_GetTemperature):
(...)
```

osarbishei, 2016-02-25:
That works! We just need one more elif for the rotation packet under the motion engine subsystem.

acourt, 2016-02-25:
oops, forgot rotation. I'm guessing there is going to be a rotation packet every time a rotation is detected?

osarbishei, 2016-02-25:
Yes. That is correct.

acourt, 2016-02-25:
[image]

acourt, 2016-02-25:
[image]

Got some accel readings.

osarbishei, 2016-02-25:
Nice! Looks good. I forgot to say that we would set the accelerometer range to 16g instead of 2g. Can you do that on both boards? It will use the same 16g the next time we power it on and it won't get saturated so easily.

osarbishei, 2016-02-25:
Just try the command: setAccFullScale 16

acourt, 2016-02-26:
Ah SHIT, matplotlib doesnt work well on the laptop. fuuuuuu

acourt, 2016-02-26:
OK, got it to work on the laptop. I'm gonna call it a night an maybe work on it tomorrow on site. I will put the scale to 16g before going to bed.

osarbishei, 2016-02-26:
Thanks Alex!

closed 2016-03-08 by acourt