Motsai/neblina-python #68: Indentation Error in /neblina-python/examples/gui/neblinastream.py

makrandgupta, 2016-03-20:
Looks like you forgot to include the commands for the second if condition in the PlottingData class. 

I got this error when trying to execute the gui example

`
/neblina-python/examples/gui$` python neblinastream.py
  File "neblinastream.py", line 68
    else:
    ^
`

acourt, 2016-03-20:
The neblinastream.py example is non-functional for now and should either be fixed or removed entirely from the public API. 

The structure of the API has changed so much since I have written it that I am considering deleting it completely and starting from scratch.

souellet-ms, 2016-06-06:
neblinastream.py, and numerous other examples which are no longer related to current Api have been removed.

closed 2016-06-06 by souellet-ms